R0rt1z2/kaeru #7: Help porting kaeru to a new device

ramabondanp, 2025-06-15:
I tried porting Kaeru to my Infinix HOT 20 5G with the MT6833 chipset.

After comparing with Ghidra, the result is:

CONFIG_APP_ADDRESS=0x4822DCC4
CONFIG_APP_CALLER=0x4822B278
CONFIG_BOOTLOADER_BASE=0x48200000
CONFIG_BOOTLOADER_SIZE=0xF2A34
CONFIG_BOOTMODE_ADDRESS=0x482F5598
CONFIG_FASTBOOT_FAIL_ADDRESS=0x4822E67C
CONFIG_FASTBOOT_INFO_ADDRESS=0x4822E5D4
CONFIG_FASTBOOT_OKAY_ADDRESS=0x4822E858
CONFIG_FASTBOOT_PUBLISH_ADDRESS=0x4822E260
CONFIG_FASTBOOT_REGISTER_ADDRESS=0x4822E224
CONFIG_LK_LOG_STORE_ADDRESS=0x4824F66C
CONFIG_MTK_DETECT_KEY_ADDRESS=0x48204614
CONFIG_PLATFORM_INIT_ADDRESS=0x482026B0
CONFIG_PLATFORM_INIT_CALLER=0x48229AB4
CONFIG_VIDEO_PRINTF_ADDRESS=0x4823AE90
CONFIG_INFINIX_X666=y
CONFIG_MEDIATEK_MT6833=y

However, Fastboot is currently not detected in Windows (showing as an unknown device), and the device enters a boot loop after a few seconds. It cannot access the home screen or recovery mode.

[lk_X666.zip](https://github.com/user-attachments/files/20745127/lk_X666.zip)

R0rt1z2, 2025-06-15:
Everything seems correct except the platform init caller, which points to an invalid address.

Can you try this image? It's a ZIP file. Please extract it and flash the LK inside:  [X666-kaeru.bin.zip](https://github.com/user-attachments/files/20745295/X666-kaeru.bin.zip)

ramabondanp, 2025-06-15:
> Everything seems correct except the platform init caller, which points to an invalid address.
> 
> Can you try this image? It's a ZIP file. Please extract it and flash the LK inside: [X666-kaeru.bin.zip](https://github.com/user-attachments/files/20745295/X666-kaeru.bin.zip)

[image]
Thank you! It's working now. Could you please tell me the correct address?

R0rt1z2, 2025-06-15:
> [image] Thank you! It's working now. Could you please tell me the correct address?

This is the full config file I've used:
```
CONFIG_APP_ADDRESS=0x4822DCC4
CONFIG_APP_CALLER=0x4822B278
CONFIG_BOOTLOADER_BASE=0x48200000
CONFIG_BOOTLOADER_SIZE=0xF2A34
CONFIG_BOOTMODE_ADDRESS=0x482F5598
CONFIG_FASTBOOT_FAIL_ADDRESS=0x4822E67C
CONFIG_FASTBOOT_INFO_ADDRESS=0x4822E5D4
CONFIG_FASTBOOT_OKAY_ADDRESS=0x4822E858
CONFIG_FASTBOOT_PUBLISH_ADDRESS=0x4822E260
CONFIG_FASTBOOT_REGISTER_ADDRESS=0x4822E224
CONFIG_LK_LOG_STORE_ADDRESS=0x4824F66C
CONFIG_MTK_DETECT_KEY_ADDRESS=0x48204614
CONFIG_PLATFORM_INIT_ADDRESS=0x482026B0
CONFIG_PLATFORM_INIT_CALLER=0x48229B62
CONFIG_VIDEO_PRINTF_ADDRESS=0x4823AE90
CONFIG_INFINIX_X666=y
CONFIG_MEDIATEK_MT6833=y
```

Feel free to open a PR to add support for the device and SoC if you'd like. Otherwise, I'm happy to take care of it myself and will credit you as the author with git. Whatever works best for you.

I assume you've already figured this out, but just in case: UART0 is at `0x11002014` and UART1 is at `0x11002000`, in case you decide to push it yourself.

If you have any other questions, feel free to ask, I'll be happy to help.

ramabondanp, 2025-06-15:
Thanks, I’ll do that later. The main reason I’m doing this is to make the volume key enter Fastboot mode. 
And now the issue on my Infinix device is that using set_bootmode(BOOTMODE_FASTBOOT) takes me to Factory Mode instead of Fastboot. Is there a way to detect the correct one?

R0rt1z2, 2025-06-15:
> Thanks, I’ll do that later. The main reason I’m doing this is to make the volume key enter Fastboot mode. And now the issue on my Infinix device is that using set_bootmode(BOOTMODE_FASTBOOT) takes me to Factory Mode instead of Fastboot. Is there a way to detect the correct one?

I just checked, and the boot mode appears to be `99` for your device as well, so I doubt that value is the issue. I feel like the key values are different on this platform/device. I can't tell from LK, since it seems to use pointers that reference these values.

Can you add this to late_init and let me know what values it shows?
```c
video_printf("Key at 0x4841e980: 0x%04x\n", *(unsigned short*)0x4841e980);
video_printf("Key at 0x4841e984: 0x%04x\n", *(unsigned short*)0x4841e984);
````

R0rt1z2, 2025-06-15:
> > ....In the stock pressing Volume Down does nothing, it just boots straight into Android OS.
> 
> Okay, I was wrong. In the stock LK, pressing the volume down key takes it to Factory Mode. My bad.

Your volume down key seems to be `0`. Place this in `late_init()` and see if it works:
```c
if (mtk_detect_key(0)) {                                                   
    set_bootmode(BOOTMODE_FASTBOOT);              
    show_bootmode(BOOTMODE_FASTBOOT);                                  
}
```

R0rt1z2, 2025-06-15:
> > > ....In the stock pressing Volume Down does nothing, it just boots straight into Android OS.
> > 
> > 
> > Okay, I was wrong. In the stock LK, pressing the volume down key takes it to Factory Mode. My bad.
> 
> Your volume down key seems to be `0`. Place this in `late_init()` and see if it works:
> 
> if (mtk_detect_key(0)) {                                                   
>     set_bootmode(BOOTMODE_FASTBOOT);              
>     show_bootmode(BOOTMODE_FASTBOOT);                                  
> }

> > If that does not work, try with `17` as well.
> 
> Great! 0 didn't work, but 17 did.

Awesome, my bad then. I guess `17` is your volume down key and `0` is volume up.

For the orange state warning and the DM-Verity warning, you can use the following code:

```c
// Suppresses the bootloader unlock warning shown during boot on
// unlocked devices. In addition to the visual warning, it also
// introduces an unnecessary 5-second delay.
// 
// This patch get rid of the delay and the warning by forcing the
// function that holds the logic to always return 0 and therefore
// not executing the code that shows the warning.
addr = SEARCH_PATTERN(LK_START, LK_END, 0xB508, 0x4B0E, 0x447B);
if (addr) {
    printf("Found orange_state_warning at 0x%08X\n", addr);
    FORCE_RETURN(addr, 0);
}

// Disables the warning shown during boot when the device is unlocked and
// the dm-verity state is corrupted. This behaves like the previous lock
// state warnings, visual only, with no real impact.
//
// Same approach: patch the function to always return 0.
addr = SEARCH_PATTERN(LK_START, LK_END, 0xB530, 0xB083, 0xAB02);
if (addr) {
    printf("Found dm_verity_corruption at 0x%08X\n", addr);
    FORCE_RETURN(addr, 0);
}
```

closed 2025-06-15 by ramabondanp