R0rt1z2/kaeru #9: Patch helpme oppo a5s mt6765

NCLnclNCL, 2025-06-23:
[lk.zip](https://github.com/user-attachments/files/20869640/lk.zip)
Lk source for mt6765
https://github.com/DavidBuchanan314/lk-mt6765

R0rt1z2, 2025-06-24:
I'm not sure what you want me to patch, since there doesn't seem to be any kind of fastboot block on this device. Either way, try the attached image and let me know if it boots: [a5s-kaeru.zip](https://github.com/user-attachments/files/20883454/a5s-kaeru.zip)

R0rt1z2, 2025-06-24:
> > I'm not sure what you want me to patch, since there doesn't seem to be any kind of fastboot block on this device. Either way, try the attached image and let me know if it boots: [a5s-kaeru.zip](https://github.com/user-attachments/files/20883454/a5s-kaeru.zip)
> 
> Have fastboot but it show black screen

Try with this one and check if you see `=> FASTBOOT mode...` when booting into fastboot mode: [a5s-kaeru.zip](https://github.com/user-attachments/files/20885698/a5s-kaeru.zip)

NCLnclNCL, 2025-06-24:
> Try with this one and check if you see `=> FASTBOOT mode...` when booting into fastboot mode: [a5s-kaeru.zip](https://github.com/user-attachments/files/20885698/a5s-kaeru.zip)

[image]
Yes, it showed

NCLnclNCL, 2025-06-24:
> Try with this one and check if you see `=> FASTBOOT mode...` when booting into fastboot mode: [a5s-kaeru.zip](https://github.com/user-attachments/files/20885698/a5s-kaeru.zip)

i want you patch when i press up volume it will boot to fastboot

R0rt1z2, 2025-06-24:
Great, give me a few minutes and I will send you another LK image to test.

R0rt1z2, 2025-06-24:
> > > > I'm not sure what you want me to patch, since there doesn't seem to be any kind of fastboot block on this device. Either way, try the attached image and let me know if it boots: [a5s-kaeru.zip](https://github.com/user-attachments/files/20883454/a5s-kaeru.zip)
> > > 
> > > 
> > > Have fastboot but it show black screen
> > 
> > 
> > Try with this one and check if you see `=> FASTBOOT mode...` when booting into fastboot mode: [a5s-kaeru.zip](https://github.com/user-attachments/files/20885698/a5s-kaeru.zip)
> 
> i want you patch when i press up volume it will boot to fastboot and remove dm verify

Try this one please: [a5s-kaeru.bin.zip](https://github.com/user-attachments/files/20886950/a5s-kaeru.bin.zip)

NCLnclNCL, 2025-06-24:
> Try this one please: [a5s-kaeru.bin.zip](https://github.com/user-attachments/files/20886950/a5s-kaeru.bin.zip)

Volume up it also boot to recovery 😂, i want it boot to fastboot

closed 2025-06-24 by R0rt1z2