R0rt1z2/kaeru #11: trying to port to a cheap walmart tablet, and getting a weird half-off state after a few seconds

Lost-Entrepreneur439, 2025-07-06:
I have an Onn Surf 3 8-inch, a cheap Walmart tablet (manufactured by Joyar, MT8168), I decided to try porting Kaeru, I followed the wiki to do everything, got it built, but the results were strange.
After flashing the kaeru partition to lk with mtkclient, I booted the tablet, it went to the Onn logo, the bootloader unlock warning message was gone, but after 10 seconds or so, the screen just turned off, and the tablet was completely unresponsive, couldn't power on, couldn't reboot, wouldn't connect to my computer, the only way I could get it out of this state was to disconnect the battery, I tried entering Fastboot to see if Kaeru was at least working, but fastboot didn't detect it, and shortly afterwards, the tablet went back to the weird half-off state. I restored the original lk and it returned back to normal, so it doesn't appear there was any damage to the tablet.

kaeru didn't have support for the 8168, so I just assumed it used the same UART values as the 8163, this is probably where the issue is coming from, but who knows, maybe I just missed smth in the wiki

I've attached my entire kaeru directory, with the broken kaeru lk (100071483-kaeru.bin) and the original one (lk.bin)

[kaeru.tar.gz](https://github.com/user-attachments/files/21084251/kaeru.tar.gz)

R0rt1z2, 2025-07-06:
Several addresses were incorrect. The main issue was that the `app()` caller had the same address as `app()` itself, which doesn't make sense, since the caller should be the address where the `blx app()` instruction resides.

Try to build with this defconfig:
```
CONFIG_APP_ADDRESS=0x56023E28
CONFIG_APP_CALLER=0x560219A0
CONFIG_BOOTLOADER_BASE=0x56000000
CONFIG_BOOTLOADER_SIZE=0x91280
CONFIG_BOOTMODE_ADDRESS=0x560940E0
CONFIG_FASTBOOT_FAIL_ADDRESS=0x56024EC0
CONFIG_FASTBOOT_INFO_ADDRESS=0x56024E40
CONFIG_FASTBOOT_OKAY_ADDRESS=0x5602508C
CONFIG_FASTBOOT_PUBLISH_ADDRESS=0x56024B1C
CONFIG_FASTBOOT_REGISTER_ADDRESS=0x56024AE0
CONFIG_LK_LOG_STORE=n
CONFIG_MTK_DETECT_KEY_ADDRESS=0x56003908
CONFIG_PLATFORM_INIT_ADDRESS=0x56002374
CONFIG_PLATFORM_INIT_CALLER=0x56020B0C
CONFIG_VIDEO_PRINTF_ADDRESS=0x5602E9BC
CONFIG_JOYAR_100071483=y
CONFIG_MEDIATEK_MT8168=y
```

And see if the device boots and if `fastboot oem kaeru-version` works.

R0rt1z2, 2025-07-07:
> > Several addresses were incorrect. The main issue was that the `app()` caller had the same address as `app()` itself, which doesn't make sense, since the caller should be the address where the `blx app()` instruction resides.
> > Try to build with this defconfig:
> > ```
> > CONFIG_APP_ADDRESS=0x56023E28
> > CONFIG_APP_CALLER=0x560219A0
> > CONFIG_BOOTLOADER_BASE=0x56000000
> > CONFIG_BOOTLOADER_SIZE=0x91280
> > CONFIG_BOOTMODE_ADDRESS=0x560940E0
> > CONFIG_FASTBOOT_FAIL_ADDRESS=0x56024EC0
> > CONFIG_FASTBOOT_INFO_ADDRESS=0x56024E40
> > CONFIG_FASTBOOT_OKAY_ADDRESS=0x5602508C
> > CONFIG_FASTBOOT_PUBLISH_ADDRESS=0x56024B1C
> > CONFIG_FASTBOOT_REGISTER_ADDRESS=0x56024AE0
> > CONFIG_LK_LOG_STORE=n
> > CONFIG_MTK_DETECT_KEY_ADDRESS=0x56003908
> > CONFIG_PLATFORM_INIT_ADDRESS=0x56002374
> > CONFIG_PLATFORM_INIT_CALLER=0x56020B0C
> > CONFIG_VIDEO_PRINTF_ADDRESS=0x5602E9BC
> > CONFIG_JOYAR_100071483=y
> > CONFIG_MEDIATEK_MT8168=y
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > And see if the device boots and if `fastboot oem kaeru-version` works. UART regs are the same as MT8163, so don't worry about that.
> 
> yup, this one worked!

Cool. For the bootloader warnings, place the following code in `board_late_init` (refer to other `board-*.c` files for guidance):
```c
uint32_t addr = 0;

// Suppresses the bootloader unlock warning shown during boot on
// unlocked devices. In addition to the visual warning, it also
// introduces an unnecessary 5-second delay.
// 
// This patch get rid of the delay and the warning by forcing the
// function that holds the logic to always return 0 and therefore
// not executing the code that shows the warning.
addr = SEARCH_PATTERN(LK_START, LK_END, 0xB508, 0x4B0E, 0x447B);
if (addr) {
    printf("Found orange_state_warning at 0x%08X\n", addr);
    FORCE_RETURN(addr, 0);
}

// Disables the warning shown during boot when the device is unlocked and
// the dm-verity state is corrupted. This behaves like the previous lock
// state warnings, visual only, with no real impact.
//
// Same approach: patch the function to always return 0.
addr = SEARCH_PATTERN(LK_START, LK_END, 0xB510, 0xB082, 0xA802);
if (addr) {
    printf("Found dm_verity_corruption at 0x%08X\n", addr);
    FORCE_RETURN(addr, 0);
}
```

Lost-Entrepreneur439, 2025-07-13:
> > > Several addresses were incorrect. The main issue was that the `app()` caller had the same address as `app()` itself, which doesn't make sense, since the caller should be the address where the `blx app()` instruction resides.
> > > Try to build with this defconfig:
> > > ```
> > > CONFIG_APP_ADDRESS=0x56023E28
> > > CONFIG_APP_CALLER=0x560219A0
> > > CONFIG_BOOTLOADER_BASE=0x56000000
> > > CONFIG_BOOTLOADER_SIZE=0x91280
> > > CONFIG_BOOTMODE_ADDRESS=0x560940E0
> > > CONFIG_FASTBOOT_FAIL_ADDRESS=0x56024EC0
> > > CONFIG_FASTBOOT_INFO_ADDRESS=0x56024E40
> > > CONFIG_FASTBOOT_OKAY_ADDRESS=0x5602508C
> > > CONFIG_FASTBOOT_PUBLISH_ADDRESS=0x56024B1C
> > > CONFIG_FASTBOOT_REGISTER_ADDRESS=0x56024AE0
> > > CONFIG_LK_LOG_STORE=n
> > > CONFIG_MTK_DETECT_KEY_ADDRESS=0x56003908
> > > CONFIG_PLATFORM_INIT_ADDRESS=0x56002374
> > > CONFIG_PLATFORM_INIT_CALLER=0x56020B0C
> > > CONFIG_VIDEO_PRINTF_ADDRESS=0x5602E9BC
> > > CONFIG_JOYAR_100071483=y
> > > CONFIG_MEDIATEK_MT8168=y
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > And see if the device boots and if `fastboot oem kaeru-version` works. UART regs are the same as MT8163, so don't worry about that.
> > 
> > 
> > yup, this one worked!
> 
> Cool. For the bootloader warnings, place the following code in `board_late_init` (refer to other `board-*.c` files for guidance):
> 
> uint32_t addr = 0;
> 
> // Suppresses the bootloader unlock warning shown during boot on
> // unlocked devices. In addition to the visual warning, it also
> // introduces an unnecessary 5-second delay.
> // 
> // This patch get rid of the delay and the warning by forcing the
> // function that holds the logic to always return 0 and therefore
> // not executing the code that shows the warning.
> addr = SEARCH_PATTERN(LK_START, LK_END, 0xB508, 0x4B0E, 0x447B);
> if (addr) {
>     printf("Found orange_state_warning at 0x%08X\n", addr);
>     FORCE_RETURN(addr, 0);
> }
> 
> // Disables the warning shown during boot when the device is unlocked and
> // the dm-verity state is corrupted. This behaves like the previous lock
> // state warnings, visual only, with no real impact.
> //
> // Same approach: patch the function to always return 0.
> addr = SEARCH_PATTERN(LK_START, LK_END, 0xB510, 0xB082, 0xA802);
> if (addr) {
>     printf("Found dm_verity_corruption at 0x%08X\n", addr);
>     FORCE_RETURN(addr, 0);
> }
> 
> If that works, feel free to open a PR and I'll merge the device support. Otherwise, I can push it manually with your authorship, whatever works best for you.

PR opened. I'll go ahead and close this issue now.

By the way, how do you get the UART regs? I have another device I want to port to (nuu-NUU_A1, the Insulet OmniPod Dash PDM/Nuu A1+), but that's MT6580, which also doesn't have the UART regs.

closed 2025-07-13 by Lost-Entrepreneur439