R0rt1z2/kaeru #17: help with porting to Insulet OmniPod Dash PDM (MT6580): Almost every address missing from generated config

Lost-Entrepreneur439, 2025-08-27:
this is a medical device that uses MT6580, i've verified that every partition seems to be unverified except for seccfg (any seccfg change causes it to start bootlooping -- lk, boot, recovery, system and all other partitions can be modified), however, compared to most other devices, almost all of the addresses are not in the generated config, and i honestly have no clue how to find any of the addresses asides from the ones mentioned in the wiki (e.g. app, platform_init), im not sure if this device has something weird going on with the lk or what, do you have any clues here? i've attached both the autogenerated config and the lk.bin dumped with mtkclient

[NUU_A1_defconfig.txt](https://github.com/user-attachments/files/22000383/NUU_A1_defconfig.txt)

[lk.bin.zip](https://github.com/user-attachments/files/22000385/lk.bin.zip)

shomykohai, 2025-08-27:
@Lost-Entrepreneur439 here you go

[NUU_A1_defconfig.txt](https://github.com/user-attachments/files/22003482/NUU_A1_defconfig.txt)

Lost-Entrepreneur439, 2025-08-27:
> [@Lost-Entrepreneur439](https://github.com/Lost-Entrepreneur439) here you go
> 
> [NUU_A1_defconfig.txt](https://github.com/user-attachments/files/22003482/NUU_A1_defconfig.txt)

tried, this one isn't working, screen doesnt power on and no computer connection when powered on (you have to remove battery then plug it in while mtkclient is open to reflash the original lk)

shomykohai, 2025-08-27:
> tried, this one isn't working, screen doesnt power on and no computer connection when powered on (you have to remove battery then plug it in while mtkclient is open to reflash the original lk)

Sorry, it seems like the fastboot_fail and fastboot_info addresses are wrong, and the first is actually pointing to something related to the logo partition.

The other addresses are correct

shomykohai, 2025-08-27:
@Lost-Entrepreneur439 fastboot_fail seem to be wrapped in your lk, to print to both screen and terminal.

I'll leave the addresses here:
`fastboot_fail_wrapped` -> `0x81e1929c` (This is what some functions seem to use)
`fastboot_fail` -> `0x81e177dc` (This is what you should use)

Lost-Entrepreneur439, 2025-08-28:
> > > > nope, still fails with that updated defconfig, same issue
> > > 
> > > 
> > > Weird indeed, i just rechecked all the addresses and they look fine to me. I wouldn't say the issue is the payload size too.
> > > You said all partitions except seccfg can be modified, so I assume the issue is not the sec policy.
> > > Did you confirm 100% that lk can be modified? I'm gonna check the sec policy to be sure as well
> > > Edit: Can you grab expdb logs?
> > 
> > 
> > [expdb.bin.zip](https://github.com/user-attachments/files/22016082/expdb.bin.zip)
> > Confirmed that LK isn't verified. Doing an edit to change the "FASTBOOT MODE" text to "SLOWBOOT MODE", the PDM still boots perfectly.
> 
> Thanks. It seems like your device isn't actually logging to expdb, weird.. It will be much harder to figure out what's actually happening unless you can as a last resort access UART for getting preloader and lk logs. Can you send the patched lk with kaeru as well? I encountered a similar issue on `penangf` after the payload got too big, so perhaps it could be a similar issue.

[NUU_A1-kaeru.bin.zip](https://github.com/user-attachments/files/22034655/NUU_A1-kaeru.bin.zip)

R0rt1z2, 2025-09-03:
I’m just as clueless as shomyand not entirely sure what’s going on. I recently realized that the way I’ve been injecting kaeru isn’t the most effective, and I suspect that might be causing issues.

The only suggestion I have is to try manually finding a better injection point by identifying unused functions in LK and seeing if that makes a difference.

shomykohai, 2025-09-13:
@Lost-Entrepreneur439 [this commit](https://github.com/R0rt1z2/kaeru/commit/64d65d351af1e39844f692276f689ce30c2d2f8d) added support for specifying a custom Injection Address.
If you can find a good injection point, you might be able to fix the issue. It is confirmed that is what was causing the same issue on penangf.

shomykohai, 2025-09-21:
@Lost-Entrepreneur439 can you try building with this in def config and see if it boots?
`CONFIG_FORCE_INJECT_ADDR=0x81e53b64`

Lost-Entrepreneur439, 2025-09-22:
> [@Lost-Entrepreneur439](https://github.com/Lost-Entrepreneur439) can you try building with this in def config and see if it boots? `CONFIG_FORCE_INJECT_ADDR=0x81e53b64`

still doesn't boot, same issue.